bigtreetech/BIQU-B1 #4: Can you control the Hot End RGB LED via GCODE?

poldim, 2020-09-21:
I know it can be turned on via the screen, but can you do it via gcode?

thisiskeithb, 2020-09-23:
Yes. See https://marlinfw.org/docs/gcode/M150.html.

LeSuedois, 2020-10-14:
Hello,
I tried the M150 command light all led's in white (`M150 B100 R100 U100`). But only the left led is white, the right one is green.
any idea how to properly light them up? 
looks like a little jitter I know from neopixels at the end of the signal line which are not properly addressed.
Thank you!

blueeagle69, 2020-11-06:
In my Marlin firmware I changed the following code;
#define NEOPIXEL_PIXELS 2
//#define NEOPIXEL_IS_SEQUENTIAL (disable this)
You should then be able to control both LED's using the M150 command.
Also the M150 command has an index switch (I) you can use also.
https://marlinfw.org/docs/gcode/M150.html

kevorm, 2023-03-01:
did you try M150 RUB  that should be all white.  unless you have a white led configured, then it would be M150 W