bigtreetech/BIQU-B1 #9: illegal flash app

pentabene, 2020-11-01:
after updating to the latest tft firmware I got "illegal flash app" twice after starting the printer (immediately, no bisque boot screen)
had to reflash the tft.

anybody else with this problem?

Ejay-3D, 2020-11-07:
You have remaining files from previous flash on your sd Card . 
You need 1 to reflash the screen , then 2 remove any files from the firmware flash ( including hidden files ._* on Mac os ) from the SD card .

pentabene, 2020-11-07:
I did that. But strangely it also happened without sd card insterted.

Ejay-3D, 2020-11-07:
Yes as i Saïd you need to reflash first to get rid of thé message

pentabene, 2021-02-18:
So just to close this for good here is what you have to do in case you have the same issue as me:

Format your SD-Card as FAT32 (even if it is already FAT32, reformat it)
Refresh the Screen Firmware, you might even have to use an older Version if it does not work.
After the Update finishes take the SD-Card out of the TFT Slot and put it in your Computer. It should show the Firmware file wit .CUR in the end instead of .bin

**Now the most important Step:**

Format your SD-Card again as FAT32 **BEFORE** you put it back into the printer. It seems erasing them sometimes doesn't suffice, so I'd suggest to format it.

closed 2021-02-18 by pentabene