bigtreetech/BIQU-B1 #10: please publish schematics for the board with the USB-C connector in the hotend

FotoFieber, 2020-11-06:
Hi

I like to design a swappable hotend and need the schematic for the USB-C board on the hotend. Would be nice, if I don't have to reverse engineer... :+1: 

Thx,
Markus

BlueM92, 2020-12-16:
I second this, I just want to make it smaller!

janf228, 2021-03-25:
@FotoFieber, @BlueM92 did you get any schmematic? Or done some reverse engineering?

Thx,
Jan

FotoFieber, 2021-04-24:
Use at your own risk and check it twice... :)
24V: A1, B1, A12, B12 (which is usually GND on USBC)
GND: A4, B4, A9, B9 (which is usually V+ on USBC)
TH+: A3, B3, A10, B10
Neopixel Data: A8, B8
CNC Fan-: A6, A7, B6, B7
Heater-: A2, B2, A11, B11
X Min: A5, B5

There is a step down converter for the neopixel on the hotend pcb.

janf228, 2021-04-25:
@FotoFieber Thank you, is the usb-c cable custom or a stock one?

GrahamJB, 2021-04-25:
The 'C' cable uses USB style connectors but is not a USB cable, definitely custom.

QuangGB, 2026-08-27:
If I have a breakout board that converts a USB 3.1 Type-C port to 24 through-hole pins, along with a 24-conductor shielded cable, I could certainly build a replacement cable myself, right?