bigtreetech/BIQU-B1 #11: Wrong nozzle position when running Z Offset

Ejay-3D, 2020-11-19:
Bug Description
When Running Z Offset The nozzle doesn't place itself at center of the bed ( Where the probe made its probe.
Instead it goes more back left .

My Configurations
B1 ABL / B1 ABL HIGH RES Branch .

Steps to Reproduce
Go To ABL
Click on Z Offset to turn it on .
Expected behavior:

Nozzles should be positioned at center of the bed when adjusting Z Offset

Actual behavior: [What actually happens]
Nozzles is to on the back and on the left of the bed ( Not in the center )

Additional Information

[video-1603372394.mp4.zip](https://github.com/bigtreetech/BIQU-B1/files/5566535/video-1603372394.mp4.zip)

Bearjejedv, 2021-11-25:
Did you find a solution to this issue?

Arnold021874, 2021-11-25:
Hi!
As of the moment, I do not experience yet that problem. I have a B1 already
but I did not yet assemble it. I highly hope, upon assembly , that I will
not be experiencing major problems. But I am confident that I will get
whatever support I might need.

Arnold F. Torres