bigtreetech/BIQU-B1 #13: No Hot END RGB

tomich, 2020-12-11:
My Hotend RGB was working fine (I don't use it that much) but since a few days ago it starts on blue and stays there. I cant change it nor turn it off. I don't know what i've changed in my cfg or in my hot end because I unclogged my nozzle and tighten the Heatbreak screws because they were loose (disassembling the hot end enclosure to do this) and I flashed 9999 different configs  with latest 2.0-bugfix to try to fix an issue with my bltouch (after all tests, BLTouch was faulty and I'm getting a replacement). 

So... I don't know what i touched or changed but I cannot control hot end LEDs and cannot turn them off. 

Can someone give me a basic round of things I can try?

I already have #define NEOPIXEL_LED enabled.

I tried forcing the port with  #define NEOPIXEL_PIN    P1_24

Attached my cfg. My changes are in lines ending with // TOMTOM

[Configuration.h.txt](https://github.com/bigtreetech/BIQU-B1/files/5676871/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIQU-B1/files/5676872/Configuration_adv.h.txt)

jfrovito, 2021-09-07:
Hi, I have the same problem. I think it happend after downgrading from latest firmware (2021-04-28) to the previous one (2020-08-18). I had to do this because I experienced many issues in last version (all of them reported in here)...
Could you fix this problem?

RowdyRocket, 2023-01-05:
The problem is that the firmware listed for the STM32 version is not the B1 firmware. The correct firmware for the Biqu B1 TFT35 is BIGTREE_TFT35_V3.0_B1.27.x.bin and can be found here...

[https://github.com/.../Copy%20to%20SD%20Card%20root...](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update?fbclid=IwAR2TXqMdH7EK0tMIsQ9_RiHe_vWpRjNHxQ-7L1JJ5vIL0xRqz0nceVTCvWo)

Grab the correct .bin file, language, config, then choose from the various theme folders and grab the TFT35 folder from the one you want, place them in your favorite SDcard and install. Should solve your troubles.

LuizMaia1981, 2024-06-17:
Seu problema não é de Firmware, e sim um problema no circuito de redução de voltagem, na placa hotmode, vou tentar achar uma solução para pelo menos desligar os leds que ficam acesos direto, se conseguir eu postarei aqui