bigtreetech/BIQU-B1 #19: E0 doesnt work

htcheroportugal, 2021-01-30:
Hi

I have an SKR 1.4 on biqu b1 printer. I've installed silent TMC2225 driver (UART mode) in X, Y, Z axis (this works) but motor of extrudor E0 not works.

I already tested motor E0 in X axis TMC2225 Driver (it works) in Z axis (it also works).

I thought that the problem was the E0 pins, so i changed them with E1 pins. This didn't work so the problem continues.
I verify all the wires but nothing happened.

I'd really appreciated your help on this.
Thanks in advanced, looking forward for your orientation

Bug Description

Expected behavior: It supposed to work.

Actual behavior: It doesn't work.

guigmachado, 2021-07-19:
Hi !!! did you resolve the issue !! i'm with the same here... build it today and the e0 dont move in the tests, changing with the X and moving it on screem the motor works,,,

htcheroportugal, 2021-07-20:
Yes, i solved.
it was problem with bad conections, on wires.
i had just some press on the wires and its got ok.

aguinaldoabbj, 2022-06-29:
Hi sorry to dig up this issue almost a year after.
I've just received my Biqu B1 and I noticed a similar problem. The extruder motor only works when it wants. It's an intermittent problem that is driving me crazy. The wiring is ok (double, triple checked) and the cables are loose. Sometimes the printer behaves normally, I can print stuff. Half an hour later, it just stops printing because the extruder doesn't move. Was your problem indeed bad conections?

guigmachado, 2022-07-04:
Hi ! it was bad connections in my case, your case seens to be something in
the software, try to update. I did this few days ago and i managed almost
all printer issues.

aguinaldoabbj, 2022-07-05:
Hi @guigmachado. Now I tested individually all stepper motors and drivers, they are all OK. What kind of software problem you are referring to? I've tested all firmware available here. Now I'm back to the stock firmware. Which version are you using?

guigmachado, 2022-07-09:
Hi, i had some issues with CURA, i dont know why but something didn't work
well with stock BIQU profile. Changing some aspects  the printer worked
well. Change the firmware is a nightmare, good luck :)

aguinaldoabbj, 2022-07-10:
> Hi, i had some issues with CURA, i dont know why but something didn't work well with stock BIQU profile. Changing some aspects the printer worked well. Change the firmware is a nightmare, good luck :) Em ter., 5 de jul. de 2022 às 16:37, Aguinaldo Bezerra < ***@***.***> escreveu:
> […](#)
> -- Guilherme G. Machado CANAC 128966

So could you please share your Cura profile for Biqu B1?