bigtreetech/BIQU-B1 #20: Entering M28 gcode command on TFT command console send printer into error.

emklap, 2021-02-02:
When uploading file to SD card through esp-01 wifi module send printer into error.
This can be reproduced by entering  the command "M28 test.gco" in the command xx of the printer.

An empty file is created on the sd card with file size zero.
Than the printer start reporting error messages

This is the reponse of the printer

M28 test.gco
echo:Now fresh file: test.gco
Writing to file: test.gco
//action:notification test.gco
ok
Error:No Checksum with line number, Last Line: 0
Resend: 1
Error:No Checksum with line number, Last Line: 0
Resend: 1
Error:No Checksum with line number, Last Line: 0
Resend: 1

The fw of the main board and TFT is the latest one available form this git hub branch.
[image]

emklap, 2021-02-03:
The problem seem to be in interfaceCmd.c, line 208, found by luc-github in hiis buig fix for ESP3D , issue 158

> I found the bug :
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/interfaceCmd.c#L208
> 
> This block every commands from WebUI but let all commands from TFT when stop polling is enabled which is wrong
> 
>   if (!ispolling && !fromTFT)
>   { //ignore any query from TFT
>     purgeLastCmd();
>     return;
>   }

See :
https://github.com/luc-github/ESP3D-WEBUI/issues/158

Can you verify this and provide a bugfix FW for the B1 TFT? I cant build it myself as the FW source code for the B1 machine is not released on github yet.

DaStivi, 2021-02-03:
@emklap sure you can build your own firmware for the b1 tft... look at my other post an linking to the btt tft github... its in there allready...  i can also sendyou an compiled firmware if you like...

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1601

definitly the wrong place here for this issue and i'm almost sure no one is monitoring this :D

emklap, 2021-02-03:
Hi @ DaStiva, thanks for pointing this out to me. This is a lack of knowledge from my part, I am completely new to 3d printing and this is my first wee of compiling printer code and I am learning fast.

I will compile the TFT code myself, that should not be a problem. I prefer to have control over my own source code so I know which changes I have made.