bigtreetech/BIQU-B1 #21: Bed leveling not enebled

stelioc66, 2021-02-28:
Hi  My printer BIQU B1 when it  starts gives me  error  (Bed leveling not enabled)  then  i get  M117 ABL is enabled.

i don't  have  BL TOUCH  or anything for auto bed leveling .

i have updated the firmware botht TFT and Printer.

Any  Help?

Thank you

Stelios

ricardo777, 2021-02-28:
I think it is a slicer setting. Maybe you have custom g code in the slicer or set bed leveling on somewhere.

stelioc66, 2021-02-28:
it does it when i start the printer
not when i start printing

chet9651, 2021-03-09:
I have got this problem but It’s okay now.
Try this: 
1. Press “Menu”
2. Press “Setting “
3. Press “Feature “
           Then find the item.    “Auto Save Load Leveling “    And disable it(red dot show behind the item)
   After that reboot.

chet9651, 2021-03-09:
) Hi My printer BIQU B1 when it starts gives me error (Bed leveling not enabled) then i get M117 ABL is enabled.

> 
> i don't have BL TOUCH or anything for auto bed leveling .
> 
> i have updated the firmware botht TFT and Printer.
> 
> Any Help?
> 
> Thank you
> 
> Stelios

I have got this problem but It’s okay now.
Try this: 
1. Press “Menu”
2. Press “Setting “
3. Press “Feature “
           Then find the item.    “Auto Save Load Leveling “    And disable it(red dot show behind the item)
   After that reboot.

https://user-images.githubusercontent.com/80184681/110464715-12cace80-8106-11eb-8f67-0d5619c9401f.MP4

chet9651, 2021-03-15:
To

> 
> 
> > ) Hi My printer BIQU B1 when it starts gives me error (Bed leveling not enabled) then i get M117 ABL is enabled.
> > > i don't have BL TOUCH or anything for auto bed leveling .
> > > i have updated the firmware botht TFT and Printer.
> > > Any Help?
> > > Thank you
> > > Stelios
> > 
> > 
> > I have got this problem but It’s okay now.
> > Try this:
> > 
> > 1. Press “Menu”
> > 2. Press “Setting “
> > 3. Press “Feature “
> >    Then find the item.    “Auto Save Load Leveling “    And disable it(red dot show behind the item)
> >    After that reboot.
> > 
> > IMG_5229.M
> 
> Thank you for your answer but i don't have this submenu or the one i see about serial

To got this submenu you should update "Touch Screen Firmware" see link below:

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

hapklaar, 2021-04-21:
This error occurs when you update your TFT firmware and forget to change config.ini to not load ABL settings on startup. If you change `auto_load_leveling:1` to `auto_load_leveling:0`, save the config file to sd, put int in your TFT SD slot, then reset, it should be gone.

DougJoseph, 2021-04-21:
> This error occurs when you update your TFT firmware and forget to change config.ini to not load ABL settings on startup. If you change `auto_load_leveling:1` to `auto_load_leveling:0`, save the config file to sd, put int in your TFT SD slot, then reset, it should be gone.

To stelioc66:

hapklaar's answer seems like a good candidate for solving this. Can you try that, and if it solves the matter, then mark this closed with a reply?