bigtreetech/BIQU-B1 #22: Next release date?

GrahamJB, 2021-03-18:
Hi,  are there any plans to release a firmware update aimed at the BIQU-B1?

I am currently writing a guide for the printer and I see a fair number of changes and improvements have been made in the TFT firmware, but as the guide is aimed at 'normal' users I definitely am not going to include compiling the firmware.

I just don't want to finish it and then find that it is out of date a few days later.

Graham Bland

thisiskeithb, 2021-03-27:
You can find always-up-to-date B1 configs in the upstream Marlin Configurations repo ([direct link](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/BIQU)), but the last stable release of Marlin was [2.0.7.2 back in Oct. 2020](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.7.2).

2.0.8 is supposed to come out Soon<sup>TM</sup>, but that could be tomorrow, next week, next month, next...

hapklaar, 2021-04-21:
I second this. For easy of use please provide precompiled images and configuration files for both Marlin 2.0.7.2 and BTT TFT firmware v.x.x.27.

It would be great if this could be done every time a new version of either Marlin or the TFT firmware is released. The differences between B1 and original Marlin + BTT TFT versions are only minimal, so this should not be too much work.