bigtreetech/BIQU-B1 #24: biqu b1

jorrdanna, 2021-04-17:
LCD on the printer says "printer not attached," for some reason. It has stopped working altogether.

yvon3247, 2021-04-17:
Call your seller

jorrdanna, 2021-04-17:
> 
> Call your seller Envoyé de mon iPad
> […](#)

If they were at all responsive or easy to get in touch with, do you honestly believe I'd be posting here?  BTT doesn't give a shit about their  customers once they have their money it seems.  So do me a favor everyone. If you actually have something useful to say that can help with the problem I'd love to hear it. I'll thank you.  If not, then spare me your three word nonsense replies.  okay?

janf228, 2021-04-17:
I only have this message a few seconds after powering the printer, then it disappears, can you post a photo of the error?

DougJoseph, 2021-04-21:
> LCD on the printer says "printer not attached," for some reason. It has stopped working altogether.

What baud rate are you showing for communication between the touchscreen and the main board? If is incorrect it can stop them from talking to each othere.

DougJoseph, 2021-04-21:
> I only have this message a few seconds after powering the printer, then it disappears, can you post a photo of the error?

Within the first few seconds after the printer is turned on, the touchscreen is seeking to establish communications with the main board. Assuming the baud rate is acceptable and the two shake hands with each other, the message is then taken away.

jorrdanna, 2021-04-21:
yup, mine used to display that message for a few seconds and then it would go away, just like yours. Now it just sits there displaying that message.  There is no error code.  I'll attatch a pic, but it isn't going to be too enlightening I'm afraid.
[image]

jorrdanna, 2021-04-21:
> Within the first few seconds after the printer is turned on, the touchscreen is seeking to establish communications with the main board. Assuming the baud rate is acceptable and the two shake hands [...]

I don't have any way of determining the baud rate because the printer is unresponsive.

yvon3247, 2021-04-21:
Yes ok this is normal my printer is the same

jorrdanna, 2021-04-21:
> 
> Yes ok this is normal my printer is the same Envoyé de mon iPad
> […](#)

No, it isn't normal. Normal would be that it displays that message for a few seconds, and then it would go away and I could begin printing.  My printer sits there like that and stays that way. The printer is unresponsive. It displays that message as long as it is powered on. I can leave it on all day and come back, and it will still be just like it is pictured. and again. As I said in the original post. The printer has stopped working altogether.

DougJoseph, 2021-04-22:
> I don't have any way of determining the baud rate because the printer is unresponsive.

Jorrdanna, in order to try to help, please let me get a little more info. You mentioned the printer is unresponsive. There are two components that could be responsive or not, the main board, and the touchscreen. Is the touchscreen itself unresponsive? If so, the printer's main board may not be unresponsive. Or vice versa. Or both...

The touchscreen is of course not the only way to communicate with the main board, but rather is a convenient, built-in way. There can also be communication via the "back" USB port on the side (in the middle). The front USB and SD slots are on the touchscreen, but those in the middle are attached to the main board.

If your touchscreen is "responding, yet cannot make the printer work," then you can press some buttons on it to not only learn what baud rate it's attempting, but also switch baud rate to try another one.

If your touchscreen not responding at all, still your mainboard might be fine, just quietly waiting for instructions.

You likely need to do some process of elimination to find out if the mainboard is working / not, if the touchscreen is working/not, or both not.

Do you any laptop device, PC, or raspberry pi, that can connect to the mainboard via the back USB?

DougJoseph, 2021-04-22:
Two other questions: 

1. I'm assuming you still have the SD card that came with the B1. If so, have you tried a firmware re-write on the touchscreen?
2. Have you taken the bottom cover off the base, to visually inspect the wire connection between the touchscreen and the mainboard? It's only six screws, four of which hold on the rubber feet, and two more in between those.

jorrdanna, 2021-04-22:
Doug, I appreciate the effort, but I bought this so I could print things.  I've got no experience in troubleshooting electronics, or in appliance repair. My interests do not lie there.  It's been about three weeks since this happened and I contacted Biqu for support, and I've yet to hear from them.  Posting here on their github was my latest idea in trying to get them to respond.  My point  is this.  Pretend that a month ago you bought a television or a microwave or something, and it broke. No one would expect you to tear it apart and fix it yourself. No one would treat you as if you were stupid for not knowing how to do so.  The fact is that I'm not comfortable tearing this thing apart. Frankly, I'm dumbfounded that it seems to be the expectation rather than the manufacturer standing behind a product for which they charged almost $300 dollars.  Again, I appreciate the effort and your time, but I think I am done with this pile of useless junk.  I'll do more research and get a printer that works properly and has customer support for when it doesn't.

DougJoseph, 2021-04-22:
> Frankly, I'm dumbfounded that it seems to be the expectation rather than the manufacturer standing behind a product for which they charged almost $300 dollars.

I'm not with BIQU or BTT. I was just trying to help. I have no "standing" to "expect" anything of you on their behalf. 

The kind of printer that "always just works" may be out there for 8 or 10 times the sub-$300 price tag, but it may not be. The industry is still relatively in its adolescent years, and the general expectation is unless you're a major corporation hiring out the service and maintenance, you're going to need to do at least some tinkering. 

My first printer was a Snapmaker 2.0, which retails for $1800, and with the enclosure, comes to $2400. Soon after buying it, the hot-end got clogged badly. Lo and behold, they had sent some extras with the initial boxing. I had to take apart the print and replace it.

Sorry I bothered you in the midst of your rant. Was only trying to help. I'll bow out.

jorrdanna, 2021-04-22:
I'm not ranting, though that last sentence of yours kind of makes me want to.  I know you aren't Biqu, I think that's clear.  I thanked you for your time and your effort more than once.   Thanks again.

closed 2021-04-22 by jorrdanna