bigtreetech/BIQU-B1 #28: Abnormal X & Y sounds from steppers after firmware 2.0.7.x update

MattRK, 2021-05-11:
After updating the firmware on my B1 from 2.0.6.x to 2.0.7.x using both the provided firmware.bin file in this repo as well as my own compiled firmware from the source code in the B1-2.0.7.x branch from Bigtreetech's Marlin repo, my B1's stepper motors now make very strange noises when doing certain moves. I've narrowed it down to the G28 the printer performs before printing anything (But not a regular G28) as well as at various points throughout a G29. These noises also happen during regular prints but the above two movements are easier to reproduce. 

I've confirmed this is a firmware issue and NOT a machine, belt or maintenance issue. Downgrading the firmware back to 2.0.6.x resolves the issue and these noises go away. The noises come back when upgrading again to 2.0.7.x. 

It's hard to describe the sounds without hearing them. Most of the noises sound like whirring or hissing noises when the printer makes fast/long X or Y-axis moves. During the G28 prior to printing, there is a bang or knocking sound. Because this is such a weird issue, I made 3 videos so you can hear the sounds. 

1) X-axis noises during a G29 - https://imgur.com/a/JWN4ImU
2) Y-axis noises during a G29  - https://imgur.com/a/VpkxEub
3) Y-axis bangs during a G28 prior to printing- https://imgur.com/a/k86c1Vq

Again, this is not a maintenance issue. There is no issue with belts, tuning, or anything physical as far as I can tell. Marlin 2.0.6.x does not make these noises AT ALL. When you upgrade to 2.0.7.x, these noises are quite loud and common.

Lekvaran, 2021-05-13:
I have the same problem.

brunonavarro082, 2021-05-13:
Same problem here too

guicha910, 2021-06-03:
Same problem here. Does anyone find a solution ? It hurts hearing to that grinding noise....

milekitic2017, 2021-06-03:
Similar problem when press P OFFSET  the Y is abnormal loud (grinding noise).It happend only on P OFFSET calibration .

Zoaap, 2021-06-03:
Same problem here!

milekitic2017, 2021-06-03:
I am not sure but some time ago I hawe similar sound on my ender 3 there was a problem with HYBRID_THRESHOLD
Then I set all to zero and this was sulution for me.Maybe this will be helpfull .

lars-danielis, 2021-06-07:
Bad idea to set the threshold to 0.
The threshold ist set to 120 by intention. Otherwise the printer can loose steps or stall.
Better is to leave the threshold there and to avoid speeds grater then that.

rwaite2, 2021-07-14:
Anybody still seeing this? I need to pull a version of marlin but its not clear if the Biqu B1 is running exactly the upstream version that I see as a date on my printer... or if they had their own changes. I would just pull latest from marlin upstream.. but seeing this issue makes me nervous.

rwaite2, 2021-07-14:
I did notice this in the later versions of marlin upstream configuration for BIQU for 2.0.8..

`#define G26_XY_FEEDRATE         20    // (mm/s) Feedrate for G26 XY moves.
#define G26_XY_FEEDRATE_TRAVEL 100    // (mm/s) Feedrate for G26 XY travel moves.`

The TRAVEL define did not exist before that version.. that sometimes x and y go faster when traveling a further distance maybe. 

Which version of the config files did you use from upstream? I'm currently trying to build the 2.0.6 branch and match the bugfix changes. It's not really clear how the firmware binaries were build in this project

LordKarlish, 2021-07-15:
I have the same problem!

IanTWright, 2021-07-31:
I had the same issue. Compiled v2.0.9.1 with Luke's configs and have not had an issue.

cristi134, 2021-08-22:
Having the same problem, tried to load 2.0.9.1 but it seems i configured something wrong cause the printer won't start.
Can someone provide the source for a working 2.0.9 version?

rwaite2, 2021-08-24:
> Having the same problem, tried to load 2.0.9.1 but it seems i configured something wrong cause the printer won't start.
> Can someone provide the source for a working 2.0.9 version?

You should be able to pull the 2.0.9.1 tagged version from marlin's repo directly. Then you need the 2 configuration files. Marlin has a separate repo for these. In that repo.. also pull the tagged version and look for the biqu b1 configs. Copy these over the 2 config files in yoyr Marlin repo. This should work..

cristi134, 2021-08-24:
> 
> You should be able to pull the 2.0.9.1 tagged version from marlin's repo directly. Then you need the 2 configuration files. Marlin has a separate repo for these. In that repo.. also pull the tagged version and look for the biqu b1 configs. Copy these over the 2 config files in yoyr Marlin repo. This should work.. at least as a functioning printer. I never observed the issue in this ticket.

I have tried this before and received an error saying the configuration file doesn't match marlin's version.

I found a temporary solution to the problem: lowering travel speed to 100, cause that's when it occured for me.

kdavale, 2021-09-24:
These noises occur because the travel speed is wrongly set. In Cura it is preset at 150 mm/s, setting this value to 100 mm/s or less solves this.

cristi134, 2021-09-24:
> These noises occur because the travel speed is wrongly set. In Cura it is preset at 150 mm/s, setting this value to 100 mm/s or less solves this.

There's no right or wrong speed and certainly there's nothing wrong with 150, as i've used it before with old firmware and there was none of that noise

kdavale, 2021-09-24:
> > These noises occur because the travel speed is wrongly set. In Cura it is preset at 150 mm/s, setting this value to 100 mm/s or less solves this.
> 
> There's no right or wrong speed and certainly there's nothing wrong with 150, as i've used it before with old firmware and there was none of that noise

It is true that with the previous firmware it did not happen, perhaps before this setting was ignored and now not, who knows ... But the truth is that the noise is because the stepper motors are running at too high a speed. 
According to the printer's manual, "Max Printing Speed ​​" 100mm/s

cristi134, 2021-09-24:
> It is true that with the previous firmware it did not happen, perhaps before this setting was ignored and now not, who knows ... But the truth is that the noise is because the stepper motors are [...]

I think that refers strictly to print speed, not travel speed. Anyway, for most models is not much difference, it's very slow anyway because of that garbage hotend.

JimMorrison723, 2022-04-10:
Hey guys, could anyone solve the issue? i just updated everything to [2022-03-27](https://github.com/bigtreetech/BIQU-B1/tree/master/Firmware/2022-03-27) and than... i found this issue :( like afraid to print because of the terrible noise. At first i thought it was because i installed bltouch, since the auto leveling made this noise first. updated the firmware and... now it has this sound no matter what :(

Zoaap, 2022-04-10:
> Hey guys, could anyone solve the issue? i just updated everything to [2022-03-27](https://github.com/bigtreetech/BIQU-B1/tree/master/Firmware/2022-03-27) and than... i found this issue :( like [...]

These noises occur because the travel speed is wrongly set. In Cura it is preset at 150 mm/s, setting this value to 100 mm/s or less solves this. "Thx to cristi134" Or update the firmware, check the facebook group for community updates.

frombaut, 2022-04-26:
If I bring down the "DEFAULT_TRAVEL_ACCELERATION" in machine settings, the noise is gone.   For description of this setting see printscreen.
[image]

Cos19, 2022-05-27:
I'm new to 3D printing so I don't know much but changing the acceleration values to 200.00 instead of 500.00 under machine settings made a difference.

ScorpionLynx, 2022-11-24:
I've got the same issue and have been trying to change the speed settings so that I don't have the sound but when I get to about 40 mm/s which is that speed, it makes printing painfully slow. I think it has something to do with the servos locking up but I don't know where I'd fix that.

robfrei, 2022-12-20:
I too am experiencing failed prints on my Biqu B1. I am using OctoPrint and Cura Slicer. My current firmware is 2.0.6. The release notes for 2.0.6 mention that the travel speed was reduced to 120 mm/s to keep the x and y stepper motors from unlocking during printing (which is exactly the problem I have). I was still experiencing this issue on firmware 2.0.6 however. In Cura (4.12.1) , the default value for Travel Speed is 150 mm/s. I reduced this to 120 mm/s and purged all of my old gcode files in OctoPrint (to make sure I don't inadvertently use a file with old gcode; I do a lot of reprints). I am printing a problematic file now, and I can already hear a difference when the hot end does a travel (its not quite as aggressive and noisy). I'm going to experiment with a travel speed of 120 mm/s on some prints. If I still have issues, I will reduce the travel speed further to 100 mm/s as others have mentioned.

cristi134, 2022-12-20:
> 

The problem is that the firmware that comes with the printer is a mess. Settling for 120 mm/s for travel speed sounds like a poor compromise, that is extremely low and you should expect blobbing on long travels. At one point I recompiled it from scratch for Marlin 2.1 and all these problems were gone. But have since switched to klipper and never looked back, using travel speed of 500mm/s with no problems.

robfrei, 2022-12-21:
I purchased my BiQu B1 about 18 months ago, It was my initial foray into the world of 3D printing. I knew, at the time, that I would be largely on my own as the B1 community is small compared to others. I primarily use my B1 to print terrain pieces for RPG, and for that, the B1 works absolutely fine (when the prints don't fail). The print time and quality of my parts are not significantly impacted by reducing travel time from 150 mm/s to 120 mm/s or even 100 mm/s (I tested this). By the way I also reduced travel acceleration from 500 mm/s^2 to 350 mm/s^2. It's too soon to say for sure, but these adjustments do seem to have helped to help so far. There's also a chance that I'm troubleshooting in the wrong area. I've read that some B1 users are experiencing network issues. I'm using OctoPrint running on a Raspberry Pi 3B. I have a 3A power supply and OctoPrint isn't reporting any low power issues, but I've read about a variety of other power related issues that can lead to dropped network connections and thus failed prints. I have an ethernet switch on my workbench so I created a static IP address for my Pi on eth0 so I'm no longer using the Pi's WIFI connection for printing (but all of this is probably a moot point because I don't think I have any networking issues between my PC and my Pi). I haven't done any testing  on my USB cable yet, but I've printed a ton of parts using this cable. I guess I'm just hoping to get back to stable printing on my entry level B1 printer ... and someday I'll probably upgrade to a more capable machine.

lutzcosta, 2025-02-26:
> Depois de atualizar o firmware no meu B1 de 2.0.6.x para 2.0.7.x usando o arquivo firmware.bin fornecido neste repositório, bem como meu próprio firmware compilado do código-fonte no branch B1-2.0.7.x do repositório Marlin da Bigtreetech, os motores de passo do meu B1 agora fazem ruídos muito estranhos ao fazer certos movimentos. Eu reduzi para o G28 que a impressora executa antes de imprimir qualquer coisa (mas não um G28 regular), bem como em vários pontos ao longo de um G29. Esses ruídos também acontecem durante impressões regulares, mas os dois movimentos acima são mais fáceis de reproduzir.
> 
> Confirmei que este é um problema de firmware e NÃO um problema de máquina, correia ou manutenção. Fazer o downgrade do firmware de volta para 2.0.6.x resolve o problema e esses ruídos desaparecem. Os ruídos voltam ao atualizar novamente para 2.0.7.x.
> 
> É difícil descrever os sons sem ouvi-los. A maioria dos ruídos soa como zumbidos ou chiados quando a impressora faz movimentos rápidos/longos no eixo X ou Y. Durante o G28 antes da impressão, há um som de estrondo ou batida. Como esse é um problema muito estranho, fiz 3 vídeos para que você possa ouvir os sons.
> 
> 1. Ruídos do eixo X durante um G29 - https://imgur.com/a/JWN4ImU
> 2. Ruídos do eixo Y durante um G29 - https://imgur.com/a/VpkxEub
> 3. Eixo Y estrondos durante um G28 antes da impressão [- https://imgur.com/a/k86c1Vq](https://imgur.com/a/k86c1Vq)
> 
> Novamente, isso não é um problema de manutenção. Não há problemas com correias, afinação ou qualquer coisa física, até onde eu saiba. O Marlin 2.0.6.x NÃO faz esses ruídos NENHUM. Quando você atualiza para 2.0.7.x, esses ruídos são bem altos e comuns.

What is the procedure to revert to 2.0.6.x? Is it just a matter of installing it normally as if I were updating?

thanhhoangvan, 2025-03-08:
I am new. This is my first printer. I just bought it 2 days ago and it was running the old firmware 2020 and it did not have that problem. But when I updated to the latest version, it made screeching noises from both the x and y axis step motors. The z axis motor is not visible, maybe it moves less so its acceleration is not enough to make a sound. This error is very annoying.

Mategm, 2025-03-08:
> I am new. This is my first printer. I just bought it 2 days ago and it was running the old firmware 2020 and it did not have that problem. But when I updated to the latest version, it made [...]

My advice is check everything on and in the printer! Check on the motherboard every cable and connector is in their place, check the engine controllers to be in their connector.
Check all the cables inside and outside, every connector is connecting. Check every movement system to be good.