bigtreetech/BIQU-B1 #31: BIQU B1 upgrade kit

Markeydoo, 2021-07-14:
Before I purchased this product I contacted BIQU via Amazon messaging and I asked them if this was a mixing hot end. I told them that I was looking for a mixing hot end and they told me that I would be very happy with this product.
I thought I was pretty up front that I wanted a mixing hot end but as you can see in the picture this is not the ZSYong mix hotend.
[image]
BEWARE this in NOT a Mix Hot End
I am in contact now with BIQU to rectify this issue.
I will post the out come.

emiliorizzo, 2021-09-21:
Hi @Markeydoo  I bought the BIQU B2 1 month ago. It's a BIQU B1 upgraded with that kit. 
After 15 days printing  I purchased the ZSyong mixer hotend and installed it. In my opinion is not a good hotend, it's tends to clogs every time. I've just replaced the extruders with Triangle Lab's BMG clones. The results are better but no enough. The underextrusion issues remains when changes the color.

Markeydoo, 2021-09-21:
Hello Emi
I have also replaced both extrudes with dual gear. 
I have had a lot of oozing from the cooling tubes. 
I set my hot end temp to 200c and I am not getting as much clogging but I still have the oozing. 
I was told by Biqu the the mix hot end is not engineered properly. 
I am working with them to get a refund.

emiliorizzo, 2021-09-21:
Hi @Markeydoo ,  thanks for yout comments! 

I was thinking of removing the ZSyong mixer from my printer, cause it's unusable. 
After your words I have no doubts about that.
I'll try to communicate with BTT. I would to upgrade  to IDEX but it's difficult to get in my country. 
By the other hand looking at the BTT repos there is little information about the IDEX upgrade and the firmware requirements.

Markeydoo, 2021-09-22:
Hi Emi
On the IDEX upgrade, I does not mix. But you should be able to stack you colors. 
What slicing software are you using. I am using Cura server. 
If I set my infill to 100% and layer to .15 I get an ok print. That's with temp at 200c. 
My big issue is that the oozing from the cooling tubes runes the print if it is more than 8 hours. 
If you get the IDEX conversion let me know how it goes.

emiliorizzo, 2021-09-22:
Hi @Markeydoo I have given up the dream of mixing. I've been reading and apparently the 'toothpaste' effect is always in the designs of these hotends.  I've just reinstalled the ZSYong 2 in 1 .
I have an Ender3 Pro with many modifications and it works very well, my idea for the biqu is two color printing.
I 'm also using cura, printing at 200C. My tests were of 4hs maximum.
With the 2-in-1 hotend, retracting the filament 40mm when printing or feeding I have no jams.
The mixer becomes clogged when the filament sits still for a period of time. The tip of filament grows in diameter due the temperature and the leaks of the other filament inside the barrel.
I think if you always mix the filaments during printing the cloggs should be less. Anyway the results (when works) are poor. The nozzle generates a two color thread instead of mixing. That's way the printed parts alternate the colors inside and outside.

Markeydoo, 2021-09-22:
Hi Emi
I worked with BTT and they gave me a refund. 
I have not seen any reviews on the IDEX upgrade kit for the B1. 
On the mix hot end, your correct, the quality is not as good as the single color. 
I’m interested in the IDEX upgrade but I would like to see some reviews. 
You may want to contact BTT for a refund. 

Thank you
Mark