bigtreetech/BIQU-B1 #32: BIQU B1 HEATING FAILED

ijaxbr, 2021-07-18:
Hi, 
Recently, i buyed a biqu B1....and after my second print, I received the message:

" Heating Failed, system stopped! Heater_ID:0"

Is it a hardware or a software  problem?

I appreciate any help to start troubleshooting...

Thanks!

Bigsmooth68, 2021-07-18:
Hi,
Most probably hardware. Check the connectivity.
Is it before or after bed warm up ? In other word, is it bed or head themistance.

ijaxbr, 2021-07-18:
Thank you for your help!
I disassembled the extruder hub and checked the wiring....it was a little twisted...(in my opinion)...then i checked the wiring inside the printer (MB, POWER SUPLY....)
Once the cables was reconnected, the extruder warmed normally.....