bigtreetech/BIQU-B1 #42: Biqu B1 Knob Led not working

Cos19, 2022-05-26:
Just upgraded to the latest firmware for motherboard and tft screen and now the knob Led is not working. The option to change its color under Feature settings is no longer there.

marcelomirandasilva, 2022-07-07:
The same thing happens to me!

hesamg59, 2022-08-26:
me too... update to 2022-03-27 and remove knob led on menu !

RowdyRocket, 2023-01-05:
The problem is that the firmware listed for the STM32 version is not the B1 firmware. The correct firmware for the Biqu B1 TFT35 is BIGTREE_TFT35_V3.0_B1.27.x.bin and can be found here...

[https://github.com/.../Copy%20to%20SD%20Card%20root...](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update?fbclid=IwAR2TXqMdH7EK0tMIsQ9_RiHe_vWpRjNHxQ-7L1JJ5vIL0xRqz0nceVTCvWo)

Grab the correct .bin file, language, config, then choose from the various theme folders and grab the TFT35 folder from the one you want, place them in your favorite SDcard and install. Should solve your troubles.

Cos19, 2023-01-07:
I'm getting Errors using the firmware from this link. So what I did is just went back to the firmware from June 2021 and the led light on the knob is now working.

RowdyRocket, 2023-01-07:
It's a bit of a mess on the link I posted, it's mixture of all the different bin files. This list of the files needed on your sdcards...

Language Packs (folder)
TFT35 (folder from your favorite theme)
admin.txt
BIGTREE_TFT35_V3.0_B1.27.x.bin
config.ini
language_en.ini (Found in the language folder not sure if it's needed but I had an error that seemed to leave when I copied it to the SD root)

Hope this helped. It worked for me.

Cos19, 2023-01-07:
Ok. So got all the correct files. Loaded with no errors, but option for led knob is now gone again...

Cos19, 2023-01-07:
Had to delete previous reply cause I was looking in the wrong place. So I loaded all the correct files and it is working. Thank you! Before, the led knob settings were under FEATURE. With this new firmware it is under SCREEN UI. Thank you again as this was irritating me for a long time.

RowdyRocket, 2023-01-08:
Awesome! I know I was so bugged by this for the longest time. Glad we got it going!

Cos19, 2023-01-11:
Have a new problem now. If I try cancel a print, the print does stop but the nozzle and bed does not start cooling down, and the hotend just stops where it is on the print. It's supposed to lift up, move to the side and the bed supposed to move forward. Does not do that. Any ideas?

RowdyRocket, 2023-01-11:
My guess is you need to setup your cancel gcode. Check your config.ini file for the TFT and see what it says near the end for the cancel gcode.