bigtreetech/BIQU-B1 #44: Using extruder E1 instead of E0 - TMC Errors and over-extrusion

aguinaldoabbj, 2022-08-02:
Hi all,

I'm experiencing a real saga with the Biqu B1 I've purchased recently. Having a terrible time with it, as the extruder only works when it wants. (https://www.reddit.com/r/BIGTREETECH/comments/vjwpns/biqu_b1_extruder_motor_getting_stuck_successively/ )

Anyway, the only workaround that showed some effect was to swap make it use channel E1 instead of E0. On the mainboard (SKR 1.4), I moved the TMC2225 driver from channel E0 to E1 (also replicated the jumper configuration from E0 on E1, see attached diagram).

[image]

I've also moved the motor plug from E0M to E1M. In the firmware, I've just swapped the pins of E0 and E1 (as can be seen in the attached code) in the code. This is the only change I made. So the firmware continues using E0 but using the E1 pins instead. Compiled using the guide in BTT's Marlin repo here in Github. 

[image]

Once the firmware is upgraded, the printer boots but shows the error "All Low" and "TMC CONNECTION ERROR". Running M122 also points to a problem with E0:

[image]

However, I kinda can print, but the extruding behavior has changed drastically. With the same slicer configuration and same nozzle (the default 0.4mm), the printer is madly **over-extruding**, like I'm using a 0.8 nozzle with a very high flow rate. Also, the thick line is irregularly shaped. All the configurations are the default ones just after the firmware upgrade.

What could I have done wrong or am I forgetting here?

Thanks in advance for any help.

radek8, 2022-10-08:
You changed the definition of the pins to E0 and E1 that's fine, but you forgot to change the definition of the uart port E0.

[image]

aguinaldoabbj, 2022-10-10:
Yeah, I ended up swapping everything and it worked.

closed 2022-10-10 by aguinaldoabbj

fzicre, 2022-10-11:
I had the same issue with my Biqu B1 and it was the teflon tube. Make sure
it's properly connected and there is no space between the tube and the
beginning of the hot end.
Also i had to disable the filament run out detection sensor, in combination
with the tube issue gave me the same thing with extruder.