bigtreetech/BIQU-B1 #46: keine Verbindung zum Drucker

matthias-12, 2022-12-07:
Kann sein, dass ich hier noch was falsch mache, da ich mich als Neuer noch nicht richtig auskenne.
Ich entschuldige mich im voraus.

Ich habe einen BIQU-B1, der nicht funktioniert.
Weil die Y Achse zu weit fährt (rattert, der Schalter von Y funktioniert), und die Leveling Punkte gegenverkehrt angefahren werden, wollte ich die neue Firmware [Firmware-SKR-V1.4.bin) aufspielen.
Leider hat das nicht geklappt, jetzt kommt die Meldung " keine Verbindung zum Drucker".
Und......das nur auf Touch Mode, auf LCD 12864 Mode läßt es sich nicht umstellen.
Alle Parameter sind auf 0 gestellt.

Kennt jemand diesen Fehler? 
Was kann ich jetzt machen, dass der Drucker wieder normal funktioniert.
Ich bin hauptsächlich Anwender und kein Spezialist!

radek8, 2023-01-27:
Use English.
You can use Google translate.
You probably used FW with the wrong configuration.
Compile your own FW with the correct parameters. In the future, you will be able to upgrade the printer and modify the fw yourself.

matthias-12, 2023-01-27:
Hallo Radek,
ich danke Dir, der Drucker funktioniert seit gestern wieder.
der sd-Loader war defekt.

Mfg 

Matthias-12
 
From: Radek
Date: 27-01-2023 21:19
To: bigtreetech/BIQU-B1
CC: matthias-12; Author
Subject: Re: [bigtreetech/BIQU-B1] keine Verbindung zum Drucker (Issue #46)
Use English.
You can use Google translate.
You probably used FW with the wrong configuration.
Compile your own FW with the correct parameters. In the future, you will be able to upgrade the printer and modify the fw yourself.

radek8, 2023-01-27:
Close the ticket