bigtreetech/BIQU-B1 #52: Necessary firmware BIQU B1 SKR 1.4

Gintaras-sur, 2024-02-08:
Good day. Does anyone have the latest firmware update for BIQU B1 SKR V1.4?

Luigisvc, 2024-02-13:
Hi, our printer is old and I doubt BTT will release new firmwares for it. so I compiled myself a new firmware based on Marlin 2.1.2.1 with input shaping and linear advance enabled for the original stepper drivers 

I am not employee or have any connection with the company, I am only a customer,  therefore I am not responsible for any problem caused after installing this firmware. That being said, here is the link for my github. I compiled 2 versions, MBL qith manual mesh feature for bed levelling and UBL for use with Bltouch or clone.

https://github.com/Luigisvc/Biqu-B1_Custom_Firmware

Gintaras-sur, 2024-02-20:
Hello. Do you know how and where to connect the auto-off relay btt v1.2? And what needs to be changed in the firmware?

Gintaras-sur, 2024-02-20:
And maybe you have configuration.h and configuration.adv.h in your firmware?

Luigisvc, 2024-02-21:
Hi, I dont have this device, but if you want, my sources are in another repo in github
https://github.com/Luigisvc/Marlin_Sources

Look for the folder of your board and bed leveling option, last version is 2.1.2.1
Change as you wish and compile.